embassy-rs/stm32-data #10: Add stm32f1 GPIO AF maps

Dirbaio, 2021-05-01:
The F1 GPIO mapping works completely different, by "remapping" all the pins for a peripheral together. 

Invent some YAML format to represent that, and extract it from the GPIO XMLs.

DCNick3, 2021-12-28:
Copying over from #85:

> True... It seems they're all "compatible" in the sense that a given bit is either present or not present, but it never does X in one chip and Y in another chip. It's like the RCC's. In this case what we've done is "merge" all the yamls to obtain a single one with all the bits, that works for all the chips in the family. The merge_regs.py script does this.
> 
> #10 is about adding info on the YAMLs about pinouts. Something like "peripehral X with remap=0 uses pins XXX, with remap=1 uses pins YYY". Since it works differently than all the other stm32 families it'll need a new YAML format.
> 
> This info is then used in embassy-stm32 to autogenerate the trait implementations to enforce correct pins when creating an UART etc.

DCNick3, 2021-12-28:
This seems to be possible, but _may_ require API changes. Because in F1 you can't change pin functions independently, the pin trait bounds like `SclPin<T: Instance>` would not work as-is, they would rather need to be defined on tuples of pins, not on individual ones.

closed 2025-08-19 by Dirbaio