embassy-rs/stm32-data #52: I2C missing in stm32l432kc

ctron, 2021-06-25:
According to the specs, the stm32l432kc should have an I2C interface. However it seems that this is not being generated.

And maybe I just didn't understand how the process works.

Dirbaio, 2021-06-25:
It does have the peripehrals, but with no block: https://github.com/embassy-rs/stm32-data/blob/main/data/chips/STM32L432KC.yaml#L160

It's missing an entry here https://github.com/embassy-rs/stm32-data/blob/main/parse.py#L308

According to [this](https://docs.google.com/spreadsheets/d/1-R-AjYrMLL2_623G-AFN2A9THMf8FFMpFD4Kq-owPmI/edit#gid=0) L4 should be i2c_v2. If you add it, compare the registers to make sure they're the same, just in case.

Dirbaio, 2021-08-18:
This has been fixed for a while already :)

closed 2021-08-18 by Dirbaio