embassy-rs/stm32-data #110: Duplicate PLLSRC entry in RCC_CFGR register of STM32F3

VasanthakumarV, 2021-12-08:
Looks like for STM32F303xB/C, STM32F358xC, STM32F303x6/8 and STM32F328x8 Bit 15 is reserved.
For STM32F303xD/E Bit 15 and Bit 16 together are used for PLLSRC.

Reference: Page 140 of F3's [Reference manual]

[Reference manual]: https://www.st.com/resource/en/reference_manual/dm00043574-stm32f303xb-c-d-e-stm32f303x6-8-stm32f328x8-stm32f358xc-stm32f398xe-advanced-arm-based-mcus-stmicroelectronics.pdf#%5B%7B%22num%22%3A224%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C67%2C371%2Cnull%5D

https://github.com/embassy-rs/stm32-data/blob/5506d27471c7e3297450127c3279f3dab96c94f8/data/registers/rcc_f3.yaml#L607-L616

Dirbaio, 2021-12-08:
Oops, that's annoying.

One option would be to split the entire RCC yaml in two.

Another would be to only keep the 2-bit-wide field, and in Embassy side make sure to only write `0b00` or `0b10` in the chips that have less choices. Luckily it seems `0b00` and `0b10` have the same "meaning" in both variants.

VasanthakumarV, 2021-12-09:
@Dirbaio Thanks for the response, I think I will keep the 2-bit wide field and not create 2 different files.
Currently, I have manually changed the file for #109,
I will keep this Issue open, to remind me to make changes to the merge script for tackling this problem programmatically.

closed 2021-12-20 by VasanthakumarV