embassy-rs/stm32-data #120: Add LPUART support (for L4)

mro95, 2022-01-25:
Hi, I would like to begin adding support for LPUART in the HAL. 
What is the workflow to add a new peripheral to the meta-pac?

One of my issues was that the `./d extract-all` script doesn't match `LPUART` but does match `LPUART1`

Dirbaio, 2022-01-25:
> One of my issues was that the ./d extract-all script doesn't match LPUART but does match LPUART1

This is expected, it matches by peripheral name, which is LPUART1. It's not an issue, just change `block/LPUART1:` to `block/LPUART:` in the yaml afterwards.

- Run `./d extract-all LPUART1`. This'll output a bunch of yamls in `tmp/LPUART1`
- Diff them between themselves, to identify differences. The differences can either be:
   - 1: Legitimate differences between families, because there are different LPUART versions. For example, added registers/fields.
   - 2: SVD inconsistencies, like different register names for the same register
   - 3: SVD mistakes (yes, there are some)
   - 4: Missing stuff in SVDs, usually enums or doc descriptions.
- Identify how many actually-different (incompatible) versions of the peripheral exist, as we must *not* merge them. Name them v1, v2.. (if possible, by order of chip release date, see [here](https://docs.google.com/spreadsheets/d/1-R-AjYrMLL2_623G-AFN2A9THMf8FFMpFD4Kq-owPmI/edit#gid=1972450814).
- For each version, pick the "best" yaml (the one that has less enums/docs missing), place them in `data/registers/lpuart_vX.yaml`
- Add entries to [`perimap`](https://github.com/embassy-rs/stm32-data/blob/main/stm32data/__main__.py#L84)
- Regen, check `data/chips/*.yaml` has the right `block: lpuart_vX/LPUART` fields

(do not use merge-regs.py, that's only for RCC)

mro95, 2022-01-26:
Thanks for the help! As you can see, I made a PR for this.
Because those things are not documented, is it possible to add your last comment (maybe with some changed) in the readme of wiki?

Dirbaio, 2022-02-05:
Done! :)

closed 2022-02-05 by Dirbaio