embassy-rs/stm32-data #182: MCK pin is not passed through on SPI peripheral

xoviat, 2023-04-08:
Looking at the `blame`, I2S has been considered a 'fake' peripheral from the very beginning. This is fine, but for the fact that the I2S peripheral contains the MCK pin, while SPI does not. The optimal behavior (IMO) would be to add the MCK pin to the existing SPI peripheral.

Dirbaio, 2023-04-08:
Ouch, this is annoying. I'm not sure what to do. We can either add I2S as a peripheral, or add the I2C pins to the SPI peripheral. I guess it depends on:

- Are all SPI instances also I2S-capable? Or only some?
- Are the pins the same as SPI, for example I2S MCK is always the same as SPI SCK? Or are they different/extra pins?

xoviat, 2023-04-19:
> when parsing pins, make the code change "peri=I2S3 pin=MCK" to "peri=SPI3 pin=I2S_MCK"
> just some string replacements
> xoviat
> Oh I understand.
> xoviat
> Am I right in assuming that I2SX = SPIX?
> dirbaio
> xoviat
> Am I right in assuming that I2SX = SPIX?
> should be the case, yes
> I haven't seen any chip that violates that assumption
>

closed 2023-06-16 by xoviat