embassy-rs/stm32-data #194: "'__INTERRUPTS' has an invalid section specifier error" on MacOS

Balint134, 2023-06-06:
Trying to compile embassy on MacOS Ventura but unable to compile `stm32-metapac` with command `cargo +nightly build --target=thumbv7em-none-eabi --verbose` resulting in an error saying 
```
LLVM ERROR: Global variable '__INTERRUPTS' has an invalid section specifier '.vector_table.interrupts': mach-o section specifier requires a segment and section separated by a comma.
libc++abi:
error: could not compile `stm32-metapac` (lib)
```

The cargo command that fails is this (omitting path names)
```
Caused by:
  process didn't exit successfully: `/.../.rustup/toolchains/nightly-aarch64-apple-darwin/bin/rustc 
--crate-name stm32_metapac 
--edition=2021 /.../.cargo/registry/src/index.crates.io-6f17d22bba15001f/stm32-metapac-9.0.0/src/lib.rs 
--error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat 
--diagnostic-width=146 
--crate-type lib 
--emit=dep-info,metadata,link -C embed-bitcode=no -C debuginfo=2 -C split-debuginfo=unpacked 
--cfg 'feature="cortex-m-rt"' 
--cfg 'feature="default"' 
--cfg 'feature="metadata"' 
--cfg 'feature="pac"' 
--cfg 'feature="rt"' 
--cfg 'feature="stm32f405og"' -C metadata=3525cf4ba37655d7 -C extra-filename=-3525cf4ba37655d7 
--out-dir /.../target/debug/deps -L dependency=/.../target/debug/deps 
--extern cortex_m=/.../target/debug/deps/libcortex_m-daffbcf4dd1d4c88.rmeta --extern cortex_m_rt=/.../target/debug/deps/libcortex_m_rt-66231a5be1130389.rmeta 
--cap-lints allow -L /.../.cargo/registry/src/index.crates.io-6f17d22bba15001f/stm32-metapac-9.0.0/src/chips/stm32f405og -L /.../target/debug/build/cortex-m-rt-9954762eb13c70bf/out` (exit status: 101)
warning: build failed, waiting for other jobs to finish...
```

I don't really understand what's going on, the `--target` flag is completely missing. In comparison the previous command before the failure has `--target`:
```
Running `/../.rustup/toolchains/nightly-aarch64-apple-darwin/bin/rustc 
--crate-name embassy_embedded_hal 
--edition=2021 /../.cargo/git/checkouts/embassy-9312dcb0ed774b29/a9fdd46/embassy-embedded-hal/src/lib.rs 
--error-format=json 
--json=diagnostic-rendered-ansi,artifacts,future-incompat 
--diagnostic-width=146 
--crate-type lib 
--emit=dep-info,metadata,link -C embed-bitcode=no -C debuginfo=2 
--cfg 'feature="defmt"' 
--cfg 'feature="embassy-futures"' 
--cfg 'feature="embedded-hal-async"' 
--cfg 'feature="embedded-storage-async"' 
--cfg 'feature="nightly"' -C metadata=6a2cf4dbb7c88f49 -C extra-filename=-6a2cf4dbb7c88f49 
--out-dir /../../target/thumbv7em-none-eabi/debug/deps 
--target thumbv7em-none-eabi -L dependency=/../../target/thumbv7em-none-eabi/debug/deps -L dependency=/../../target/debug/deps 
--extern defmt=/../../target/thumbv7em-none-eabi/debug/deps/libdefmt-2af4f290fd631f26.rmeta 
--extern embassy_futures=/../../target/thumbv7em-none-eabi/debug/deps/libembassy_futures-f33762c7422513c9.rmeta 
--extern embassy_sync=/../../target/thumbv7em-none-eabi/debug/deps/libembassy_sync-73d6df3d07e4cec8.rmeta 
--extern embedded_hal_02=/../../target/thumbv7em-none-eabi/debug/deps/libembedded_hal-5767e02773188679.rmeta 
--extern embedded_hal_1=/../../target/thumbv7em-none-eabi/debug/deps/libembedded_hal-2157b2a669028910.rmeta 
--extern embedded_hal_async=/../../target/thumbv7em-none-eabi/debug/deps/libembedded_hal_async-8dfa8c33a24bdcd5.rmeta 
--extern embedded_storage=/../../target/thumbv7em-none-eabi/debug/deps/libembedded_storage-9047e307f281c8b3.rmeta 
--extern embedded_storage_async=/../../target/thumbv7em-none-eabi/debug/deps/libembedded_storage_async-f8910109d28d4e52.rmeta 
--extern nb=/../../target/thumbv7em-none-eabi/debug/deps/libnb-4d9de40819d25c0a.rmeta 
--cap-lints allow -L /../../target/thumbv7em-none-eabi/debug/build/defmt-d3f5690588630066/out`
```

Dirbaio, 2023-06-06:
Try adding `resolver=2` to the Cargo workspace.

Balint134, 2023-06-06:
@Dirbaio Thanks a lot, that did the trick!!