embassy-rs/stm32-data #202: The use of u8-newtypes instead of enums for some fields prevents exhaustive matching

cbiffle, 2023-06-28:
Ran into this on the stm32g0/c0 I2C block (`i2c_v2` in your language). The `ISR.DIR` flag in the peripheral is a single-bit field. I had code like this (using the svd2rust APIs):

```rust
match i2c.isr.read().dir() {
    DIR_A::Read => ...,
    DIR_A::Write => ...,
}
```

Porting over to the metapac API, I ran into an issue with the way `Dir` is defined:

```rust
pub struct Dir (pub u8) ; impl Dir { # [doc = "Write transfer, slave enters receiver mode"]
pub const WRITE : Self = Self (0) ; # [doc = "Read transfer, slave enters transmitter mode"]
pub const READ : Self = Self (0x01) ; }
```

(Side note: it looks like you're using `quote` to generate code. This makes it hard to read unless you format the results. If you would like a crate to do this, [I've written one.](https://github.com/oxidecomputer/hubris/tree/master/build/call_rustfmt))

Because `Dir` admits any u8 value, the compiler requires an exhaustive match on it to cover 0..256. But that's silly -- it's a 1-bit field, and there's no way any value above 1 can emerge from the read.

This also affects the `set_dir` call, which needs to deal with many potential bogus input values (and does so by ignoring the top 7 bits, which is not ideal).

It'd be lovely if...

- Enums for 1-bit fields generated actual two-variant enums, or more generally,
- Enums for any field where there are variants covering all `2**N` options use Rust enums instead of newtypes.

This would allow exhaustive match, and ensure that setters receive valid inputs, at least for the cases where this can be done easily without custom range-limited integer types or something.

cbiffle, 2023-06-28:
I was recently writing my own attempt at a code generator (which I'm hoping to retire in favor of yours), and I wound up having field access operations for fields with potentially undefined values generate an enum containing an `Invalid(u8)` variant (for example). That way I could log the invalid bits, but not treat them as valid, and ignoring that variant was straightforward. That being said, I don't think I love that approach because it wound up creating different enums for read (contains invalid option) and write (does not contain invalid option) -- one of svd2rust's habits I was specifically trying to fix.

What if the choice of what to generate were explicitly controlled in the definition file? i.e. instead of the generator being smart and making decisions that could seem arbitrary, you could do something like `exhaustive: true` or `complete_enum:` and it would

- Produce an exhaustive enum covering all bit patterns using the things you've defined, so that the API operates in terms of that Rust enum for both read and write, or
- Produce an error if the type you've described wouldn't actually cover everything.

In that case, if someone wanted an enum containing the `INVALID_xxx` cases, they could define one. (I hesitate to generate those automatically.) This feels like the most culturally-Rust-like option to me.

I think there are numeric fields for which your current approach is ideal -- the PLL VCO settings, for instance, or USART BRR -- where you really _do_ want to pass in an integer and the integer happens to be 8 or 16 bits. If I want a VCO divisor of 16, it's nice to pass 16 instead of, like, `PllDivisor::from_u32(16).unwrap()`. So in those cases, it'd be nice if not specifying `enum:` retained the current behavior.

Dirbaio, 2023-06-28:
> generate an enum containing an Invalid(u8) variant

I think this would generate code bloat, since converting from `u8` to the enum would have to check if the variant is valid or not. Also the enum would be 2 bytes instead of 1. I don't think the compiler will be able to optimize that out in all cases.

> What if the choice of what to generate were explicitly controlled in the definition file?

I'd like to keep the reg yamls "language-independent". Like, just describing the registers, no options for controlling codegen for a particular language. 

It also adds the burden of manually annotating the enums. Unless we make "extract YAML from SVD" automatically annotate them based on some criteria, but then what'll happen in practice is all enums will stay annotated with whatever the extraction decided, so we might as well make codegen apply the criteria instead.

Also, `chiptool` still supports the "generate straight from SVD" workflow (like svd2rust), other PACs use it (like [`rp-pac`](https://github.com/embassy-rs/rp-pac)), so the problem remains there.

cbiffle, 2023-06-28:
> I think this would generate code bloat

I largely agree.

> Also the enum would be 2 bytes instead of 1

So, your current enums for 1-bit fields are 8 bits. Normally, reading and testing a 1-bit field on ARM is a matter of doing `lsls rX, #POSITION` and then testing either the carry or negative flag. In my tests, expressions like `gpioc.idr().idr(14) == Idr::HIGH` are currently doing this, optimizing away the intermediate 8-bit value, but if you're concerned about the compiler failing to optimize away enum-related tests, this may become a problem in the future.

> I'd like to keep the reg yamls "language-independent".

Ah. After chewing on this for a long time I decided this wasn't the right goal for me -- there's simply too much difference in the metadata you need to generate a really fluent interface in (say) C vs Rust, when the type systems are so wildly different (e.g. in C all enums can contain any invalid value and there is no two-valued bool type -- even the stdbool in recent revisions can contain 2, say).

Would you be open to having some sort of "overlay" data to guide API generation in a specific language, keeping a well-defined subset of the YAMLs language-independent?

> It also adds the burden of manually annotating the enums.

Ah -- I had assumed that part of the intent of checking in the YAMLs describing things is that we could go through the files (as I'm doing now) and fix and annotate all the parts that aren't producing reasonable APIs, because the SVDs are fundamentally written for debuggers -- ST doesn't use them to generate their APIs, for instance. So to me, this burden -- adding some additional fields to guide code generation in language X or Y -- doesn't seem like a significant increase.

The SVD arguably doesn't contain enough metadata to generate a reasonable C++ API, much less Rust. The files you've checked in provide an opportunity to add more information to guide those cases.

> Also, chiptool still supports the "generate straight from SVD" workflow (like svd2rust), other PACs use it (like [rp-pac](https://github.com/embassy-rs/rp-pac)), so the problem remains there.

Well, having a reasonable default for extracted SVD APIs seems wise -- what I'm suggesting, in case it was unclear, is the ability to guide its decisions in cases where a human has visited the output to fix all the SVD's errors and omissions. Similar to adding enum definitions to cases where the SVD either doesn't have one, or has a completely wrong one (about to send you a PR for the CFGR1.MEM_MODE field on G0 for instance).

The RP2040 is an unusual case, because its SVD is written by a cooperative vendor and is reasonably correct. Perhaps they'd be happy with automatic codegen without tweaks.

Dirbaio, 2023-06-28:
> I had assumed that part of the intent of checking in the YAMLs describing things is that we could go through the files (as I'm doing now) and fix and annotate all the parts that aren't producing reasonable APIs

The intent of checking in the YAMLs is 
- allow fixing mistakes and typos in the SVDs (there are A LOT of those), without a complicated patching system.
- ensuring consistency between chips (i.e. check in a single `i2c_v2.yaml` and ensure it is good, vs trying to patch wildly differing SVDs for what's an identical IP block into being consistent). `stm32-rs` doesn't even attempt this, this is the main reason `stm32-data` was started. Writing a single HAL for all stm32 chips (`embassy-stm32`) is impossible otherwise.

The workflow is usually "extract yaml from all SVDs that have i2c_v2, diff them. The diffs alone reveal a lot of typos and mistakes and missing enums. Fix it up into a single "golden" yaml, check it in".

I'd prefer it if yamls extracted straight from SVD would directly result in good codegen (modulo SVD mistakes). Adding a step that's "go through all enums and annotate them" is quite a bit of extra work.

Do we really need the two codegen styles though? What are examples of enums that we'd *actively want* the newtype codegen and not an enum? You've mentioned
- USART BRR, but that's already not an enum
- RCC PLL enums like [this one](https://github.com/embassy-rs/stm32-data/blob/5bf71d3d89663c1b919d652ee9b2a6c0977ef726/data/registers/rcc_h7.yaml#L3591). That's the most pathological case, and even then making half the variants `INVALID_XX` wouldn't be *that* bad. If you want to treat it as an integer, it should be still quite ergonomic if we codegen'ed `from_u8`/`into_u8`. `from_u8` would mask it to 7 bits then transmute, so it'd still be fast.

Are there any others?

cbiffle, 2023-06-28:
> The intent of checking in the YAMLs is

Thanks for clarifying your intent. The merging of peripherals that are identical but have slightly divergent definitions in the SVD, so that code could potentially be written to be generic over different MCUs with the same peripheral (gasp!), or even generic over _which GPIO port they're talking to_ (gasp!!), is a feature I missed from C when using stm32-rs, so I'm very much here for it!

> I'd prefer it if yamls extracted straight from SVD would directly result in good codegen (modulo SVD mistakes). Adding a step that's "go through all enums and annotate them" is quite a bit of extra work.

One of the most common SVD mistakes I encounter is missing or crappy enumeratedValues definitions. For instance, as you've likely noticed, the official SVDs for the C0s and most of the G0s have all their enum names changed to `B_0x0` and whatnot -- the literal value rather than the name. So generally speaking I assume that, when doing anything based on an SVD, I'm going to have to go through and annotate every enum. It looks like you've already done a lot of it -- the fact that the G0/C0 peripherals are derived from previous generations that had good SVDs has made the job easier. But there will come a future generation with nonsense enum names that aren't similar to (say) the L4.

Anyway, what I'm saying is: I feel like annotating the enums is a thing that will wind up happening no matter which approach you go with.

> Do we really need the two codegen styles though?

I don't know that we do! It partially depends on how you want the APIs to look, and to some degree undefined enum variants.

- in C, an "enum" is more of a collection of integer constants, and any enum variable can contain any arbitrary integer -- so exhaustive matching always requires a `default:` in the `switch`. (There's usually a compiler warning available for forgetting it, but it's rarely on by default.) So in a C API, modeling (say) a 4-bit field containing 12 valid variants using a 12-value enum is probably quite reasonable. If you want to set it to missing variant 13, you can just do that, the type system doesn't care.
- in Rust, enums must always contain one of their variants, so in that same case you might automatically generate 4 `INVALID_xxx` variants and call it a day.

This might be good enough, particularly if you (as you mention) codegen integer conversion operations. (Can I suggest the `num_traits::FromPrimitive/ToPrimitive` traits there? It's what we use in Hubris and they come with derive macros.)

There is a pattern that this might encourage that has me a little concerned though: with `INVALID` _variants_ it's not possible to write a match that is guaranteed to match all valid variants in the face of future code changes, without listing every. invalid. variant. Which in the case of (say) the PLL VCO divisors, can be dozens.

Example: it would be tempting to write

```rust
match peripheral.register().field() {
    Field::A => ...,
    Field::B => ...,
    _ => panic!("invalid"),
}
```
However, if a `Field::C` were added (possibly on a different revision of chip) that replaced one of the INVALID cases, it would be silently treated as invalid without even a warning. Personally, I would prefer a compile error that makes me decide how to handle the case, instead of silently deciding for me.

One way you could achieve that is to bin all the invalid cases under one variant using an enum with a field, `Invalid(_)`. Then the match becomes

```rust
match peripheral.register().field() {
    Field::A => ...,
    Field::B => ...,
    Field::Invalid(x) => panic!("invalid value {x:02x}"),
}
```

You've mentioned concerns about the enum now being two bytes -- you can avoid this by using an enum for the nested type instead of (say) a `u8`. Rustc's niche optimization will do the right thing unless the sum of all valid + invalid cases exceeds 256. Writing such an enum by hand would be annoying, but, you're a code generator. :-)

So you might wind up with something like, for a 3-bit field containing two valid cases,

```rust
enum Field { A, B, Invalid(InvalidField) }
enum InvalidField { _2, _3, _4, _5, _6, _7 }
```
(names are not intended to be final!)

(Incidentally, there are a lot of cases in register field modeling where fancier enum types and Options make a lot of sense -- for instance, cases where there's a valid bit that determines whether other bits are even _defined_ is a good Option case. This is part of what led me to decide that it was important to be able to add language-specific information to the definitions, since there's no equivalent concept in C, though it can be done in C++.)

cbiffle, 2023-06-28:
(Side note: if you tried something like that nested enum approach, and wanted to confirm that the in-memory representation is what you expect, the information is available in the DWARF but can be hard to interpret. I've recently [released a tool that can show your enum types in diagram form](https://github.com/cbiffle/debugdb/blob/main/src/bin/tysh.rs) including the niche optimizations, if that's useful. It's what I use to check my embedded data structures for compactness/padding/etc.)

cbiffle, 2023-06-28:
...okay, sorry to keep replying to myself here, but:

There's also an open question of how to behave if the hardware produces a bit pattern that it's documented not to produce. You've expressed concern about boilerplate if the user is required to handle it; I agree with that concern. In many cases, if the hardware is misbehaving, the only reasonable thing to do is abort the program.

But I also agree that library-mandated panics are bad, particularly in embedded contexts.

So, the `Invalid(_)` above is an attempt to compromise between those two poles.

Shutting up now and waiting for your reply. :_)

Dirbaio, 2023-06-28:
> Can I suggest the num_traits::FromPrimitive/ToPrimitive traits there? It's what we use in Hubris and they come with derive macros

These generate `match`es. The compiler compiler should able to optimize them to a cast if the u8 value always matches the enum and the enum has all `2**n` variants, but I think it'd be better to generate the unsafe cast manually to save compile time.

> You've mentioned concerns about the enum now being two bytes -- you can avoid this by using an enum for the nested type instead of (say) a u8. Rustc's niche optimization will do the right thing [...]

We'd need to make the enum variant values match the hardware though. For this you need `#[repr(u8)]` and then niche optimization no longer works: https://play.rust-lang.org/?version=stable&mode=debug&edition=2021&gist=08f4a0ec51cc4728a10e5f824655f917

If we let Rust choose the enum variant values, they won't match the hardware, and `from_u8`/`into_u8` will have to generate bloaty `match`es.

> In many cases, if the hardware is misbehaving, the only reasonable thing to do is abort the program.

even if we're OK with mandated panics, it still has the bloat problem. If we do:

```rust
let is_bar = FROB.foo().read().bar() == vals::Bar::SOME_VARIANT;
```

it currently compiles to "read reg, extract field bits, check equality". 
If `.bar()` had to validate the read variant is fine, it'd compile to "read reg, extract field bits, do a match to see if the bits are valid, panic if not, check equality".

I don't think we should spent cycles and flash space *actively* verifying the hardware is not misbehaving...

Dirbaio, 2023-06-28:
this is the "bloat" i'm referring to: https://godbolt.org/z/EY4dr5n7Y

The only "zero cost" way I can find is to add "InvalidXX" variants to the enum itself... Unless there's some magic trick to have the `Foo/InvalidFoo` enums with niche optimization but still keep them transmutable that I'm not aware of...

Dirbaio, 2023-06-28:
> Are there any others?

Yep. registers where you have to write a magic "key" value for safety.

example from rp: https://github.com/embassy-rs/rp-pac/blob/2ca948cf46681261bf91b61b26f1499fd21d3e41/src/xosc/vals.rs
stm32s have these things too.

An enum with 65536-n "invalid" variants is not going to work, no...

Dirbaio, 2023-06-28:
:eyes:  https://github.com/embassy-rs/chiptool/commit/e21dc0ee164f3d0bdbc1ce773e8651e8118eb17c

Dirbaio, 2023-06-28:
code size diff from building all `embassy-rp` examples before and after the chiptool changes:
[image]

Some increase, some decrease, but overall it's a slight win. I think it's "zero cost" enough :D

text sum before: 2658260
text sum after: 2656720

Dirbaio, 2023-06-29:
stm32-metapac v12 has enums now :upside_down_face:

closed 2023-06-29 by Dirbaio