embassy-rs/stm32-data #203: `.set_XXX()` method shouldn't be available for `<Reg>.read()`

eZioPan, 2023-06-28:
```rust
    let rcc = stm32_metapac::RCC;
    rcc.cr().read().set_hseon(true);
```

`.set_hseon()` shouldn't be available after a `.read()` method.

Currently, this is compiled with no error, and runs on MCU without a panic.

Luckily, using `.set_XXX()` after a `.read()` doesn't change the register, but it's still confusing user.

eZioPan, 2023-06-28:
and `.read()` use a `*mut T`, which might be not strict enough?

```rust
impl<T: Copy, A: Read> Reg<T, A> {
    #[inline(always)]
    pub fn read(&self) -> T {
        // use `self.ptr as *const T` instead?
        unsafe { (self.ptr as *mut T).read_volatile() }
    }
}
```

Dirbaio, 2023-06-28:
This is an intentional design decision. `rcc.cr().read()` returns an instance of `stm32_metapac::rcc::vals::Cr`, which is "a value of the CR register". This value has getters and setters for all the fields.

It allows storing and manipulating a rregister value independently of reads and writes. For example, you can do:

```rust
let mut val = rcc.cr().read();
val.set_hseon(true);
rcc.cr().write_value(val);
```

you could also store `val` in a struct, or hav a function return a register value and then have a nother do the write...

The alternative is to do "read proxy" and "write proxy" objects like svd2rust does, but:
- it literally *doubles* compile time
- it doesn't allow storing and manipulating values of a register independently from the read/write like the above snippet.

> and .read() use a *mut T, which might be not strict enough?

As far as I know, both are the same w.r.t Rust's memory model and aliasing rules, so using always `*mut` is correct.

eZioPan, 2023-06-28:
understood!