embassy-rs/stm32-data #218: Incorrect generated flash size for STM32G030K8

Sizurka, 2023-07-05:
A [STM32G030K8](https://www.st.com/en/microcontrollers-microprocessors/stm32g030k8.html) has 64 KB flash, but the generated definitions list [32 KB](https://github.com/embassy-rs/stm32-data-generated/blob/758db05cf3a813217c4051f74a2741ce288a6026/data/chips/STM32G030K8.json#L18).  This gets propagated into the metapac definitions and ends up making the memory.x define less flash than is actually available, resulting in link failures when the resulting firmware is larger than 32 KB.

This appears to be happening because the [STM32Cubeprog definition](https://github.com/embassy-rs/stm32-data-sources/blob/ca89656bdb4c9121ca8a0777bec8934e8fae3af0/cubeprogdb/db/STM32_Prog_DB_0x466.xml#L98) has two configurations, but the second one is the one with the full flash (the 0x20 occurence) size while stm32-data-gen uses the [first unconditionally](https://github.com/embassy-rs/stm32-data/blob/931ab215e93ff31ab764a29b72e802e5e07fcdb7/stm32-data-gen/src/memory.rs#L226C29-L226C29).  Then it ends up limiting the [region size to less than the total flash size](https://github.com/embassy-rs/stm32-data/blob/931ab215e93ff31ab764a29b72e802e5e07fcdb7/stm32-data-gen/src/chips.rs#L1128C28-L1128C28) (since there's no 2nd bank).   I haven't looked to see if this is actually coming up with any other chips; this is just the one I stumbled on.

Dirbaio, 2023-07-05:
Yes, there is a lot of chips with this issue.

I think the cubeprog xmls are not a good data source for this. It's impossible to know which "configuration" is the right one without actually reading registers from a live connected chip. We have to get the data from somewhere else or enter it manually...

Dirbaio, 2023-11-01:
tracking in #301

closed 2023-11-01 by Dirbaio