embassy-rs/stm32-data #222: STM32H7 DAC is a new variant

adamgreig, 2023-07-22:
Currently there's a V1 and V2 DAC and the H7 is set to use V1. Unfortunately it seems it's really a similar but different "V3", with changes in TSEL, TEN, and BOFF, plus a new CEN field:

V1 (from STM32F4 RM0090):
[image]

STM32H745 reference manual RM0399:
[image]

Additionally the TSEL enum is changed:

V1:
[image]

STM32H745:
[image]
[image]

JuliDi, 2023-07-22:
Just as a note regarding the STM32H7: For the H7A3/B3/B0 variants, there is also RM0455 which may or may not contain differences to RM0399 (I didn't spot any for the aforementioned registers, though).

JuliDi, 2023-07-22:
The [version spreadsheet ](https://docs.google.com/spreadsheets/d/1-R-AjYrMLL2_623G-AFN2A9THMf8FFMpFD4Kq-owPmI/edit#gid=1972450814)would probably need to be corrected, as well, to reflect the "version bump" to v3.

commit 4bd67390 (2023-07-22): Merge pull request #223 from JuliDi/main

WIP: fix #222 Add DAC v3 for STM32H7
PR #223 (2023-07-22): WIP: fix #222 Add DAC v3 for STM32H7
This PR should fix #222 by adding a DAC v3 variant based on v2.

TODO:
- [ ] Test whether it works now for boards with RM0399 and RM0455
- [ ] The [version spreadsheet ](https://docs.google.com/spreadsheets/d/1-R-AjYrMLL2_623G-AFN2A9THMf8FFMpFD4Kq-owPmI/edit#gid=1972450814) needs to reflect the "version bump" to v3.
closed 2023-07-22 by Dirbaio