embassy-rs/stm32-data #236: `TIM` actually has multiple versions

diondokter, 2023-08-13:
Currently the `TIM` peripheral only has a v1 version. While newer timers still seem to have the same registers and fields, some of them have extra functionality added.

I ran into this when I wanted to set the `MMS2` field of the `CR` register of the advanced timer (`TIM1`) on the `stm32g030f6`.

Apparently this problem is already known, but I'm posting it as an issue now so that is also known in writing.

xoviat, 2023-09-11:
The solution is to add extension blocks like TIM_ADV, but for those specific extensions.

TaylorCarpenter, 2024-01-27:
@xoviat Can you explain why an extension block is preferable over having an additional peripheral version? It seems to me that the `TIM` peripheral YAMLs shouldn't have been collapsed into a single version.

eZioPan, 2024-02-05:
@TaylorCarpenter no worry, I have already working on version-ized TIM: #364 and embassy-rs/embassy#2499, most of the code should just work. Curently I try add TIM_OR[123] register into PAC, they are different in each STM32 serials, I guess it will take some time though.

TaylorCarpenter, 2024-02-05:
@eZioPan and it looks like you have the UIFCOPY related offsets in there, which is excellent. That is the piece that I am looking for as it allows a time driver to be made against timers with only a single CC register.

dlaw, 2024-04-11:
I believe this issue can be closed (due to the excellent work in https://github.com/embassy-rs/stm32-data/pull/364, merged a few weeks ago). I am now using the MMS2 field of STM32G4 TIM1 CR with the git version of stm32-metapac.

closed 2024-04-12 by Dirbaio