embassy-rs/stm32-data #258: USB FS not defined for STM32L432KB

kaylynn234, 2023-09-16:
Hi there!

I'm not 100% on whether this is the correct repository to be opening an issue in, but FS USB capabilities seem to be missing from the data definitions for the STM32L432KB, and subsequently there's no `USB_FS` interrupt or `embassy_stm32::usb::Driver::new_fs` function available downstream.

The datasheet for the L432KB (and L432KC) makes explicit mention of FS support [in the first page, as well as in section 3.30](https://www.st.com/resource/en/datasheet/stm32l432kc.pdf) so I assume this is unintentional. I've had a look through this repository and I'm not entirely sure how to go about rectifying this, else I would've opened a PR instead.

All of that being said, I might be missing something - please let me know if so!

Dirbaio, 2023-09-16:
USB is present, and the interrupt is also named just "USB": https://github.com/embassy-rs/stm32-data-generated/blob/main/data/chips/STM32L432KB.json#L2669

on the embassy-stm32 side it's also present https://docs.embassy.dev/embassy-stm32/git/stm32l432kb/peripherals/struct.USB.html

so it all should work.

kaylynn234, 2023-09-16:
Right - so that *is* full-speed USB then? Some of the examples use `USB_FS` interrupts and have a `Driver::new_fs` function with a different parameter set, so I wasn't sure

Dirbaio, 2023-09-16:
yes, that's the FS usb. there's no other usb in that chip.

kaylynn234, 2023-09-16:
Right, that makes sense. Sorry for the bother!

closed 2023-09-16 by kaylynn234