embassy-rs/stm32-data #293: STM32G474CB Missing Quad SPI peripheral

Tupelov, 2023-10-18:
For whatever reason, the generater only outputs the qspi interrupt but not the actual peripheral itself.

xoviat, 2023-10-20:
looks like we don't have an implementation for that version at all, so it will need to be added

Tupelov, 2023-10-25:
> looks like we don't have an implementation for that version at all, so it will need to be added

I've tried various times to create a v4 version of the peripheral but for whatever reason it won't show up in the output. From what I could tell, only the bit width of a single register was changed and to my knowledge they changed the name of the qspi enable register but even after modifying the RCC yaml, I couldn't get it to work.
Could there be something else besides the chips.rs and quadspi_v4.yaml that I need to edit?

dlaw, 2024-04-17:
I just used quadspi via stm32-metapac on STM32G473, so I can confirm that #311 fixed this issue and this issue can be closed.

closed 2024-04-17 by Dirbaio