embassy-rs/stm32-data #298: Dac in stm32g431c8

jrmoulton, 2023-10-30:
The dac definitions are missing from this chip. It seems like it already has an entry in the perimap though. 

It also has the data for it in the json.
https://github.com/embassy-rs/stm32-data-generated/blob/4f60a9c3c9583915a33c842a35574bead53aa8da/data/chips/STM32G431C8.json#L670

What would I need to change?

Dirbaio, 2023-10-30:
maybe it's not matching because it's named "DAC1", not "DAC". Otherwise, the actual version in the cubedb xmls is different to the ones listed in perimap.

Dirbaio, 2023-10-31:
the issue is the JSON doesn't have `registers` field. See [here](https://github.com/embassy-rs/stm32-data-generated/blob/4f60a9c3c9583915a33c842a35574bead53aa8da/data/chips/STM32G431C8.json#L670) no `registers`, vs [another chip](https://github.com/embassy-rs/stm32-data-generated/blob/4f60a9c3c9583915a33c842a35574bead53aa8da/data/chips/STM32H7A3RI.json#L759) that does have it.

There's no registers because no entry in `perimap` is matching. You'd have to debug why, and fix it.

jrmoulton, 2023-10-31:
Ah I see. Thanks

closed 2024-01-02 by jrmoulton