embassy-rs/stm32-data #301: Incorrect memory sizes tracking issue

Dirbaio, 2023-11-01:
If you've got a chip where the `stm32-metapac/memory.x` feature generates a wrong memory.x please post here.

Identified cases:

- STM32G030K8 flash generates 32kb but has 64kb
- STM32WL55JC ram generates 12kb but has 64kb

TaylorCarpenter, 2024-01-27:
STM32WB55VG ram generates 128kb but has 192kb.

phycrax, 2024-01-29:
STM32G030C8 flash generates 32kb but has 64kb

trepidacious, 2024-02-03:
stm32g491re memory.x shows 32kb RAM, chip should have 112kb (3 blocks of 80, 16, 16)

Dirbaio, 2024-02-05:
STM32F303ZE says it has 64k but only has 48kb https://github.com/embassy-rs/stm32-data-generated/issues/1
stm32u5a5zj says it has 32k but it has 2.5mb or something

eZioPan, 2024-02-07:
STM32F411RE:

| Type | memory.x | DS Tab.2  | Note |
|------| -------- | --------- | ---- |
| RAM  |   64K    |  128 KB   |      |

eZioPan, 2024-03-05:
|     Chip         |  Type  | memory.x | DS Tab.2               |        Note        |
| :--------------: | :----: | :------: | :--------------------: | :----------------: |
|  STM32H5[67]xxG  |  RAM   |  256 K   |  640 (256+64+320) KB   |                    |
|  STM32H5[67]xxI  |  RAM   |  256 K   |  640 (256+64+320) KB   |                    |

Dirbaio, 2024-03-12:
|     Chip    |  Type  | memory.x | actual  |        Note        |
| :---------: | :----: | :------: | :-------: | :----------------: |
| STM32F051R8T6 |  RAM   |  7 K   |  8 KB   |   #437         |

LorenzoCucchi, 2024-03-15:
|     Chip    | Type | memory.x | actual | Note |
|:-----------:|:----:|:--------:|:------:|:----:|
| STM32F756ZG |  RAM |   320K   |  256K  |      |

64 K less since DTCM starting at 0x20000000 is not used, and RAM starts at 0x20010000.

adri326, 2024-03-22:
|     Chip    |  Type  | memory.x | DS Tab.2  |        Note        |
| :---------: | :----: | :------: | :-------: | :----------------: |
| STM32L4A6ZG |  RAM   |  320K   |  256K   |                    |

Opposite issue to the one above: `SRAM` correctly starts at `0x2000_0000` up through `0x2004_0000`, but the last `64K` aren't available. According to the doc, there's a second `SRAM2` memory device with `64K` size, mapped *both* to `0x1000_0000` (which is present in the generated `STM32L4A6ZG.json`, although it has size 0), and to `0x2004_0000`

spcan, 2024-03-25:
STM32F746ZG variant presents the same issue as the STM32F756 variant described before in this thread.

| Chip         | Type     | memory.x | Actual | Note
|--------------|-----------|:------------:|:--------:|--------|
| STM32F746ZG | RAM      |  320K        |  256K |    |

64K less because DTCM should not be counted. RAM starts at 0x20010000 not at 0x20000000.

Consider separating SRAM1 (240K) and SRAM2 (16K).

wagcampbell, 2024-04-11:
As mentioned in the Matrix chat, seeing some incorrect sizes for the STM32U5 Family.

From the reference manual (RM0456):

Flash configuration:
RM0456, 7.3.1, Table 51., 52., 53.:
[image]
[image]
[image]

SRAM configuration:
RM0456, 6.3.1, Table 45
[image]

Also, from the product page:
[image]
[image]
[image]
[image]
[image]
[image]

I haven't studied other STM32 MCUs this closely, but is it unusual to have so many SRAM devices?
> Up to seven SRAMs are embedded in the devices

I'm working on a fix, but thought I'd post this before getting ahead of myself.

Dirbaio, 2024-06-08:
https://github.com/embassy-rs/stm32-data/commit/3eab5e1161312efc394d2a238ba8b93e174846a1 fixes all issues reported here and more. I believe all stm32 memory maps are correct now.

commit 3eab5e11 (2024-06-08): memory: many memory map fixes.

fixes #301
closed 2024-06-08 by Dirbaio