embassy-rs/stm32-data #352: DacDma trait implementations for DACs on g4 are missing

jrmoulton, 2024-01-04:
I'm trying to find the data source for where DacDma implementations are for the DACs are coming from but I can't find where that is coming from in the data sources. 

https://docs.embassy.dev/embassy-stm32/git/stm32l475re/peripherals/struct.DAC1.html#impl-DacDma1%3CDAC1%3E-for-DMA1_CH3

This is what I have been looking for with grep after I downloaded all of the sources but I'm not finding it. 

@adamgreig I saw you pushed through the rework for DACs. 
Any idea where I should be looking?

adamgreig, 2024-01-05:
In the course of checking why this wasn't working I found the error and opened #353 to fix it, it looks like fundamentally the problem is that ST's CubeMX database calls them "DAC1_CHANNEL1" for G4 and "DAC1_CH1" for everything else, so embassy-stm32's build script doesn't pick them up.

jrmoulton, 2024-01-05:
Ahh thanks. Much appreciated

PR #353 (2024-01-05): Rename G4 DMAMUX DACn_CHANNELm to DACn_CHm to match all other families.
This is required for embassy-stm32 to generate the DacDma trait impls too.

Closes #352.
commit 14a87eb6 (2024-01-05): Rename G4 DMAMUX DACn_CHANNELm to DACn_CHm to match all other families.

Closes #352.
closed 2024-01-05 by Dirbaio