embassy-rs/stm32-data #459: PR #455 is breaking SPI and CRC in embassy-stm32 for stm32h7

taunusflieger, 2024-04-08:
The changes to the CRC and SPI registers in https://github.com/embassy-rs/stm32-data/pull/455 are braking changes for embassy-stm32 at least for stm32h7 ...

```
   Compiling stm32-metapac v15.0.0 (/Users/michael/src/work/stm32-data/build/stm32-metapac)
   Compiling embassy-stm32 v0.1.0 (/Users/michael/src/work/embassy/embassy-stm32)
error[E0599]: no method named `dr` found for struct `stm32_metapac::crc::Crc` in the current scope
   --> src/crc/v2v3.rs:139:17
    |
139 |         PAC_CRC.dr().read()
    |                 ^^ help: there is a method with a similar name: `cr`

error[E0599]: no method named `dr` found for struct `stm32_metapac::crc::Crc` in the current scope
   --> src/crc/v2v3.rs:147:17
    |
147 |         PAC_CRC.dr().read()
    |                 ^^ help: there is a method with a similar name: `cr`

error[E0599]: no method named `dr` found for struct `stm32_metapac::crc::Crc` in the current scope
   --> src/crc/v2v3.rs:152:17
    |
152 |         PAC_CRC.dr().read()
    |                 ^^ help: there is a method with a similar name: `cr`

error[E0599]: no method named `dr` found for struct `stm32_metapac::crc::Crc` in the current scope
   --> src/crc/v2v3.rs:159:17
    |
159 |         PAC_CRC.dr().read()
    |                 ^^ help: there is a method with a similar name: `cr`

error[E0599]: no method named `dr` found for struct `stm32_metapac::crc::Crc` in the current scope
   --> src/crc/v2v3.rs:163:17
    |
163 |         PAC_CRC.dr().write_value(word as u32);
    |                 ^^ help: there is a method with a similar name: `cr`

error[E0599]: no method named `dr` found for struct `stm32_metapac::crc::Crc` in the current scope
   --> src/crc/v2v3.rs:164:17
    |
164 |         PAC_CRC.dr().read()
    |                 ^^ help: there is a method with a similar name: `cr`

error[E0599]: no method named `dr` found for struct `stm32_metapac::crc::Crc` in the current scope
   --> src/crc/v2v3.rs:169:21
    |
169 |             PAC_CRC.dr().write_value(*word as u32);
    |                     ^^ help: there is a method with a similar name: `cr`

error[E0599]: no method named `dr` found for struct `stm32_metapac::crc::Crc` in the current scope
   --> src/crc/v2v3.rs:171:17
    |
171 |         PAC_CRC.dr().read()
    |                 ^^ help: there is a method with a similar name: `cr`

error[E0599]: no method named `txdr` found for reference `&stm32_metapac::spi::Spi` in the current scope
   --> src/spi/mod.rs:741:23
    |
741 |         let dr = self.txdr();
    |                       ^^^^ help: there is a method with a similar name: `txdr8`

error[E0599]: no method named `rxdr` found for reference `&stm32_metapac::spi::Spi` in the current scope
   --> src/spi/mod.rs:749:23
    |
749 |         let dr = self.rxdr();
    |                       ^^^^ help: there is a method with a similar name: `rxdr8`

error[E0599]: no method named `rxdr` found for struct `stm32_metapac::spi::Spi` in the current scope
   --> src/spi/mod.rs:822:22
    |
822 |         let _ = regs.rxdr().read();
    |                      ^^^^ help: there is a method with a similar name: `rxdr8`

For more information about this error, try `rustc --explain E0599`.
error: could not compile `embassy-stm32` (lib) due to 11 previous errors
```

Dirbaio, 2024-04-08:
Yes, that PR is intentionally a breaking change. embassy-stm32 needs to be updated.

dlaw, 2024-04-17:
Can be closed now that https://github.com/embassy-rs/embassy/pull/2791 and https://github.com/embassy-rs/embassy/pull/2828 are merged

closed 2024-04-17 by Dirbaio