embassy-rs/stm32-data #479: Suspicious Memory layout for STM32G4

AdinAck, 2024-05-13:
As outlined in #478 , the SRAM size for these chips looks wrong. But I am not sure how to correct it.

[image]

Dirbaio, 2024-05-14:
If the ram is aliased, just put it twice. Something like this should work:

`CCM_SRAM_ICODE 0x10000000 10, SRAM1 0x20000000 16, SRAM2 0x20004000 6, CCM_SRAM_DCODE 0x20005800 10`

The memory.x generator will use the largest contiguous block, in this case 32kb stating at 0x20000000.

AdinAck, 2024-05-14:
Gotcha, on it.

AdinAck, 2024-05-14:
I have amended and reopened #478 . I don't have the other variants of the G4 so I can't test the new memory layouts and as such I did not fix those.

AdinAck, 2024-05-14:
[image]

Surely this is a typo in the G441xB datasheet 🤣

closed 2024-05-14 by Dirbaio