embassy-rs/stm32-data #504: STM32L452: Vrefintcal address is unaligned

esden, 2024-07-17:
The address of the Vrefintcal is at `0x1fff_75aa..0x1fff_75ab` (16bit value) the issue is that this address is not 32bit word aligned, so it can't be read using the regular embassy pac functions, as it results in a panic.

Additionally the `0x1fff_75a8..0x1fff_75a9` address contains the `TS_CAL1` value. This is one of the two factory temperature sensor calibration values. (they are currently not exposed by the pac as far as I understand)

I am not sure how to address the issue. If the address is moved to be 32bit word aligned the vrefintcal value we are interested in is in the upper 16bits, which might need some handling too.

For reference the locations of these values are documented in the datasheet https://www.st.com/resource/en/datasheet/stm32l452cc.pdf on page 40.

Dirbaio, 2024-07-17:
you can make 16-bit registers with `bit_size: 16`, those emit LDRH/STRH, it's fine to make them not 32-bit aligned.

esden, 2024-07-17:
Ok, in that case. It seems that the register itself has the correct size but not the register block: https://github.com/embassy-rs/stm32-data-generated/blob/main/stm32-metapac/src/registers/vrefintcal_v1.rs#L22

Will it be enough to add `bit_size: 16` to the block section here? https://github.com/embassy-rs/stm32-data/blob/main/data/registers/vrefintcal_v1.yaml#L6

If that is all that needs to be done I will create the appropriate PR.

Dirbaio, 2024-07-17:
yep, that should work.

esden, 2024-07-17:
Ok, I started looking into making the change.

Adding the 16bit width does not seem to make any difference.

```diff
diff --git a/data/registers/vrefintcal_v1.yaml b/data/registers/vrefintcal_v1.yaml
index a40c0ddc0..e56fa7738 100644
--- a/data/registers/vrefintcal_v1.yaml
+++ b/data/registers/vrefintcal_v1.yaml
@@ -4,6 +4,7 @@ block/VREFINTCAL:
   - name: DATA
     description: Factory calibration
     byte_offset: 0
+    bit_size: 16
     access: Read
     fieldset: DATA
 fieldset/DATA:
```

The problem seems to be with the `.read()` on the `.data()`, that in turn uses `read_volatile` on a non aligned pointer.

Here is the actual panic: `ERROR panic: PanicInfo { message: unsafe precondition(s) violated: ptr::read_volatile requires that the pointer argument is aligned and non-null, location: Location { file: "library/core/src/panicking.rs", line: 221, col: 5 }, can_unwind: false, force_no_backtrace: false }`

commit 5c4497ad (2024-07-18): vrefint: Remove fieldset as it is not really needed here and causes read align issues.

Fixes #504
PR #505 (2024-07-18): vrefint: Remove fieldset
The fieldset forces a 32bit word align during read. This causes a panic on devices where the vrefint value is stored at a 16bit word and not 32bit word boundry.

Fixes #504
closed 2024-07-18 by Dirbaio