embassy-rs/stm32-data #522: Handling multiple simultaneous revisions of peripheral (ADC on U5)

klownfish, 2024-09-17:
I’m having trouble implementing the ADCs for the U5 series due to it having multiple simultaneous ADC revisions. Some chips have up to three ADC instances: ADC1, ADC2, and ADC4. While ADC1 and ADC2 are identical, ADC4 is a completely different revision. I haven't seen a setup like this before so I'm unsure on how to best handle it. 

The most straightforward solution in my opinion is to treat them as separate peripherals (e.g. `adc12_u5.yaml` and `adc4_u5.yaml`), but this breaks the pattern of having a generic "adc" peripheral across all chips.

An alternative would be to use a single file with separate blocks, similar to how timers are handled. However due to overlapping fields and enums everything related to ADC4 would need to be prefixed (e.g. `ADC4_xxxx`)

Are there any existing examples of this kind of situation, or is there a better way to approach it? Any guidance or feedback would be greatly appreciated! 

Thanks!

taylorh140, 2024-09-25:
I think I'm running into this same problem trying to include the FMPI2c peripherals on the stm32f446 and a few others:
 - the other I2C's on the device are "I2c_v1" type
 - But the FMPI2C is "I2c_v2" type
 
 and i get a fun error:
 
[image]

klownfish, 2024-09-25:
Yeah this seems to be the same core problem that I ran into. There's also this issue that's similar I realized just now
https://github.com/embassy-rs/stm32-data/issues/509