embassy-rs/stm32-data #528: Windows Builds seems to be broken

taylorh140, 2024-09-27:
To reproduce grab the present main and build on a window machine.
[image]
You should get this error. It has something to do with the fmt ? maybe? 

STATUS_STACK_BUFFER_OVERRUN

seems bad.

taylorh140, 2024-09-29:
So I took the folder from windows after d.ps1 executed and moved it to my linux setup. And got the same problem. So this is actually probably a problem with the d.ps1 and maybe not with the rust code.

brunob45, 2024-10-07:
./d.ps1 is pulling embassy-rs/stm32-data-sources@74b97817d4c4ed0db9d19a8eac46720b3c5b0d57 (April 12) but

./d is pulling embassy-rs/stm32-data-sources@709c1060ac2ec57f6042f2b4eb9cf8c1821a6c57 (April 29)

Manually pulling embassy-rs/stm32-data-sources@709c1060ac2ec57f6042f2b4eb9cf8c1821a6c57 or embassy-rs/stm32-data-sources@1a3751550575f8ffee5e45307713a3e08bc9ffb4 (Jun 7) fixes the build.

taylorh140, 2024-10-08:
This is good advice.

closed 2024-10-08 by taylorh140