embassy-rs/stm32-data #531: STM32G4 dual flash bank configuration

u6bkep, 2024-10-22:
I was having some trouble with the flash erase routines in the STM32g4 HAL, so I did some digging. The first issue I cam across was missing register definitions, so I fixed those. [pull request incoming](https://github.com/u6bkep/stm32-data/commit/3d3605c9f9e7239e4629f2d58523334275294cac). Then I noticed the G4, category 3 devices at least, default to dual bank mode meaning any attempt to erase any data in the second half of ROM will fail because the page offset is calculated incorrectly. I have confirmed erasing a page in bank 2 can be done by running the erase sequence with manual register writes.
[image]

now I need some advice on how to move forward with this chip. Is there a recommended way in the HAL to configure single bank mode so the erase function works? or is the correct solution to modify the the [metadata](https://github.com/embassy-rs/stm32-data-generated/blob/400b02bed6679118b8b78bd219ba00273fae5bf8/stm32-metapac/src/chips/stm32g474re/metadata.rs) to reflect the two memory banks?

Dirbaio, 2024-10-22:
The metadata and the embassy-stm32 implementation are for single-bank mode.

The problem is the firmware itself can't switch the chip from dual to single bank mode since it shuffles the flash, corrupting the firmware. You have to set it beforehand before flashing. I think ST's tools can do it, maybe probe-rs can also do it by writing the right value by flashing to that option byte address. 

I think the only "fix" we can do in embassy-stm32 is to detect the chip is in dual-bank mode and panic, with a message for the user.

Dual-bank G4s are a bit cursed because there's a *hole* between the banks (unless you have the biggest flash size, 512kb). So you can't pretend they're a single bigger contiguous flash region. Erase size also changes: 2kb in dual bank, 4kb in single bank. So i'd recommend sticking to single bank if you can.

zhihu2andy, 2026-06-03:
I am wondering if the issue had been fixed. I still experienced the same issue on STM32G473CB chip with 128 KB flash and there is a 384 KB hole between 2 banks. I want to erase the last 2 pages (2*2048 KB) in the the flash.
Erase last 2 pages fails with dbg message print out.
 0.000122 [TRACE] Erasing from 0x804f000 to 0x8050000 (<crate> └─ <invalid location: defmt frame-index: 14>:0)
0.000152 [TRACE] Erasing sector: FlashSector { bank: Bank2, index_in_bank: 30, start: 134541312, size: 2048 } (<crate> └─ <invalid location: defmt frame-index: 15>:0)
0.022216 [TRACE] Erasing sector: FlashSector { bank: Bank2, index_in_bank: 31, start: 134543360, size: 2048 } (<crate> └─ <invalid location: defmt frame-index: 15>:0)

supportprimecouirer-sketch, 2026-08-19:
> I am wondering if the issue had been fixed. I still experienced the same issue on STM32G473CB chip with 128 KB flash and there is a 384 KB hole between 2 banks. I want to erase the last 2 pages [...]

Can you help me my friend I need to flash like two different banks can you help me and if all goes well I will show you what it mean to work with me because I always need bank flashing and crypto flashing always

supportprimecouirer-sketch, 2026-08-19:
Can you help me my friend I need to flash like two different banks can you help me and if all goes well I will show you what it mean to work with me because I always need bank flashing and crypto flashing always

zhihu2andy, 2026-08-19:
I never tried the latest fix and don't   know if the bug is fixed. What I did is using low-level register configuration to make it work. The below is the code.
pub async fn erase_app_nvm_page() {
    let mut reg: u32;
    let mut bit_mask: u32;

    unsafe {
        // unlock the FLASH control register
        const FLASH_KEYR_REG_ADDR: *mut u32 = 0x4002_2008 as *mut u32;
        reg = 0x4567_0123;
        core::ptr::write_volatile(FLASH_KEYR_REG_ADDR, reg);

        Timer::after_micros(10).await;

        reg = 0xCDEF_89AB;
        core::ptr::write_volatile(FLASH_KEYR_REG_ADDR, reg);

        Timer::after_micros(10).await;

        // Wait for any ongoing operations to finish
        const FLASH_SR_REG_ADDR: *mut u32 = 0x4002_2010 as *mut u32;
        reg = core::ptr::read_volatile(FLASH_SR_REG_ADDR);
        bit_mask = 1 << 16;
        while reg & bit_mask !=0 {
            reg = core::ptr::read_volatile(FLASH_SR_REG_ADDR);
            bit_mask = 1 << 16;
        }

        //clear all errors
        reg |= 1 << 0;  // EOP
        reg |= 1 << 1;  // OPERR
        reg |= 1 << 3;  // PROGERR
        reg |= 1 << 4;  // WRPERR
        reg |= 1 << 5;  // PGAERR
        reg |= 1 << 6;  // SIZERR
        reg |= 1 << 7;  // PGSERR
        core::ptr::write_volatile(FLASH_SR_REG_ADDR, reg);

        Timer::after_micros(10).await;

        // 2. Erase Page 63 on Bank 2
        const FLASH_CR_REG_ADDR: *mut u32 = 0x4002_2014 as *mut u32;
        reg = core::ptr::read_volatile(FLASH_CR_REG_ADDR);
        reg |= 1 << 1; // PER[1]: set page erase bit
        reg &= !(1 << 11); // BKER[11]: set bank 1
        reg &= !(0x7f << 3); // PNB[9:3]: Clear previous page bits
        reg |= 63 << 3; // page 63 or last page of flash
        reg |= 1 << 16; // start[16]: start erase op
        core::ptr::write_volatile(FLASH_CR_REG_ADDR, reg);

        Timer::after_micros(10).await;

        reg = core::ptr::read_volatile(FLASH_SR_REG_ADDR);
        bit_mask = 1 << 16;
        while reg & bit_mask !=0 {
            reg = core::ptr::read_volatile(FLASH_SR_REG_ADDR);
            bit_mask = 1 << 16;
        }

        reg = core::ptr::read_volatile(FLASH_CR_REG_ADDR);
        reg &= !(1 << 1);
        reg &= !(0x7f << 3);
        reg |= 1 << 31; // lck[31]: set lock bit
        core::ptr::write_volatile(FLASH_CR_REG_ADDR, reg);

        Timer::after_micros(10).await;
    }
}

supportprimecouirer-sketch, 2026-08-19:
> Can you help me my friend I need to flash like two different banks can you help me and if all goes well I will show you what it mean to work with me because I always need bank flashing and crypto [...]

I can't do flashing this is the bank details and amount First receiver details and amount is 
€500,000,000 five hundred million euros

zhihu2andy, 2026-08-19:
You need set the bank mode to "single-bank" instead of "dual-bank". The code I shared with you should work.