embassy-rs/stm32-data #553: Wrong memory map entry for RAM_D3 for a STM32H747XI

dhartig, 2024-12-31:
The mapping for the RAM_D3 segment for STM32H747XI is currently mapped at address:
 &nbsp;&nbsp;&nbsp;&nbsp;`0x1800000`
according to [this document ](https://www.st.com/resource/en/reference_manual/rm0399-stm32h745755-and-stm32h747757-advanced-armbased-32bit-mcus-stmicroelectronics.pdf)(figures below) it should be in the SRAM4 area at 
 &nbsp;&nbsp;&nbsp;&nbsp;`0x38000000`

This applies for both the STM32H745/755 and STM32H747/757 series MCU

I encountered this while running [probe-rs](https://github.com/probe-rs/probe-rs) which had a problem flashing until I corrected this memory map. 

I will create a PR with the 1 character change suggested.

FYI: Here is the memory map for this series of processors

[image]
[image]

dhartig, 2024-12-31:
proposed fix in PR [#554](https://github.com/embassy-rs/stm32-data/pull/554)

Dirbaio, 2025-01-01:
fixed in https://github.com/embassy-rs/stm32-data/pull/554

closed 2025-01-01 by Dirbaio