embassy-rs/stm32-data #560: Is there a guide to adding alternate pins?

saosebastiao, 2025-01-04:
I'm currently trying to write a firmware for an existing board which uses alternate function pins which are not mapped in the PAC for embassy-stm32. Is there any guidance on how those alternate functions could be added to this repo?

Dirbaio, 2025-01-04:
the data comes from XMLs like [this one](https://github.com/embassy-rs/stm32-data-sources/blob/main/cubedb/mcu/IP/GPIO-STM32U59x_gpio_v1_0_Modes.xml) that are provided by ST as part of STM32CubeMX, and parsed by [this code](https://github.com/embassy-rs/stm32-data/blob/main/stm32-data-gen/src/gpio_af.rs).

which AF is missing? I'd start by checking the XML to see if it's in some "special" format that `gpio_af.rs` is missing. if it's really missing from the XMLs it'd need adding `if`s to the rust code to add exceptions. That shouldn't happen tho, the XMLs have been quite complete so far.

saosebastiao, 2025-01-06:
I don't have anything exhaustive, I was just trying to generate an embassy-rs SPI3 instance for an STM32G474CE, and it would not accept pin PC12 as a SPI3_MOSI pin, which the data sheet lists as being available under Alternate Function 6 (see page 75/236.). 

```
            let spi = spi::Spi::new(
                p.SPI3, 
                p.PC10, 
                p.PC12, // This causes error, but PB5 does not.  
                p.PC11, 
                p.DMA1_CH5, 
                p.DMA1_CH6, 
                spi3_config    
            );
```

[stm32g474ce.pdf](https://github.com/user-attachments/files/18325313/stm32g474ce.pdf)

However, it does appear to be present in the XMLs https://github.com/embassy-rs/stm32-data-sources/blob/1a3751550575f8ffee5e45307713a3e08bc9ffb4/cubedb/mcu/IP/GPIO-STM32G47x_gpio_v1_0_Modes.xml#L3337-L3359, so I'm not exactly sure how it would end not working with the compiled code.

Dirbaio, 2025-01-07:
STM32G474CE only exists in two packages STM32G474CETx (LQFP48), STM32G474CEUx (UFQFPN48), both of which don't have the PC12 pin. So it's correct that you can't use PC12 for SPI3_MOSI, because PC12 doesn't exist at all. 

However, due to *reasons*, AFs for non-existing pins are filtered out, but the pins themselves aren't, so `p.PC12` still exists in `embassy-stm32`. I can see how that's confusing :smiling_face_with_tear: .

saosebastiao, 2025-01-07:
Oof, that's a very interesting gotcha. In a sense, the type system did an amazing job, cause it caught a perfectly valid and fairly expensive potential error (I was using the wrong microcontroller!), albeit with a root cause that was pretty hard to understand. Thanks for the help, I really appreciate it :)

closed 2025-01-07 by saosebastiao

Dirbaio, 2025-01-07:
the ghost pins were bothering me so I fixed them: https://github.com/embassy-rs/stm32-data/commit/2624ca6d995b2570c64c6bc16fd15c76b2668605 and https://github.com/embassy-rs/embassy/pull/3738 change codegen so non-existing pins are completely gone now, so `p.PC12` doesn't exist at all now,

saosebastiao, 2025-01-07:
Absolutely amazing, thank you so much!