embassy-rs/stm32-data #568: STM32C0xx family missing ADC module.

Cydget, 2025-02-14:
I'm not sure if this is a simple change and overlooked, or if it is significantly different from all other ADC modules and thus needs its own chip specific driver. I don't believe I have the technical know how on how to add support myself, but I'm willing to learn if anyone is willing to help offer guidance.

Cydget, 2025-02-14:
Looking at  https://github.com/STMicroelectronics/stm32c0xx-hal-driver/blob/main/Src/stm32c0xx_hal_adc.c  and comparing it to https://github.com/STMicroelectronics/stm32g0xx-hal-driver/blob/master/Src/stm32g0xx_hal_adc.c I see almost no differences, so I think it would be V5, but thats just a guess. I haven't started following the "how to add peripheral" steps from the read me yet. I'm still at the point where I want to see how much work it would be to add it stage before I dig in deeper. 

Looking at the excel tracker it looks like the c0xx is pretty new and hasnt had much work done on adding it yet. https://docs.google.com/spreadsheets/d/1-R-AjYrMLL2_623G-AFN2A9THMf8FFMpFD4Kq-owPmI/edit?gid=1972450814#gid=1972450814

Cydget, 2025-02-14:
Ok, after following the read me steps, I've confirmed the /ADC/c011.yaml and /ADC/C031.yaml are identical. Additionally the c011.yaml is only different from /ADC/g0c1 by the following(excluding enum)
`Added in c011.yaml

AWD2CH19
AWD2CH20
AWD2CH21
AWD2CH22

AWD3H19
AWD3H20
AWD3H21
AWD3H22

CHSEL19
CHSEL20
CHSEL21
CHSEL22

SMPSEL19
SMPSEL20
SMPSEL21
SMPSEL22

VBATEN <- does not exist on C011 but does on g0c1

`
From this, I'm thinking the c0xx family is the same as g0xx, but with a few more channels. 

I've tried modifying /perimap.rs 
by adding lines 114 and 110 

[image]
But when I go to generate with

xoviat, 2026-11-06:
This is resolved: commit c57bc677f ("Added STM32C0 ADC data.") is in main. STM32C0 ADCs are now mapped to the shared G0 ADC peripheral (`stm32-data-gen/src/perimap.rs`: `STM32C0.*:ADC` → `adc_v2_g0`), so generated C0 chips (e.g. stm32c011f6) include `pub mod adc` pointing at `adc_v2_g0.rs`. Closing as completed.

closed 2026-11-06 by xoviat