embassy-rs/stm32-data #574: Feeling stupid, how would I add a custom board

Alan01252, 2025-03-04:
I'm trying to get started with embassry and rust development.

I have an existing program that I want to try porting over to embassay, but I am stuck at the first hurdle. 

The other project uses stm32duino, and I have a custom board like this

# Generic F030RCTx
GenF0.menu.pnum.GENERIC_F030RCTX=Generic F030RCTx
GenF0.menu.pnum.GENERIC_F030RCTX.upload.maximum_size=262144
GenF0.menu.pnum.GENERIC_F030RCTX.upload.maximum_data_size=32768
GenF0.menu.pnum.GENERIC_F030RCTX.build.board=GENERIC_F030RCTX
GenF0.menu.pnum.GENERIC_F030RCTX.build.product_line=STM32F030xC
GenF0.menu.pnum.GENERIC_F030RCTX.build.variant=STM32F0xx/F030RCT

But I don't believe this board is supported by this stm32-data repository as I can't find a matching board number, and somethin with the same pins.

I tried to follow the readme to try and understand how I would add this board, but am completed baffled.

Anyone with a bigger brain able to please fill me in with what I might be missing here to get started with this board?

Thanks

Alan

ninjasource, 2025-03-11:
Hi Alan,

Welcome! I would start with finding out the exact mcu on your board. Embassy supports mcu's, not boards to the best of my knowledge. It is up to you to use the correct pins for your specific board. Take a look at the embassy examples for inspiration.  The product line above says `STM32F030xC`. If you look at the `embassy\embassy-stm32` crate you will see all the supported chips that are exposed as features (see `embassy-stm32\Cargo.toml`. For example, I can see the that `stm32f030rc` chip is supported so that may work for you.

> "I tried to follow the readme to try and understand how I would add this board, but am completed baffled."
This crate is for adding support for new mcu's and the built-in (inside the mcu) peripherals they support, not for custom boards that use those mcus, that's probably why.

I hope that helps!

David

Alan01252, 2025-03-11:
Hi David!

Thank you, this actually works absolutely perfectly, and the rust environment once I'd figured this out is an absolutely dream to program and debug in compared to our CMAKE/Stm32duino environment.

Everything "just worked", absolutely amazing.

Thank you!

closed 2025-03-11 by Alan01252

ninjasource, 2025-03-11:
I can't take credit but I'm glad it worked for you! Some great people working on this project.