embassy-rs/stm32-data #672: Enable full duplex I2S on STM32F405RG?  Without MCK?

bifurcation, 2025-09-21:
The [STM32F405RG datasheet](https://www.st.com/en/microcontrollers-microprocessors/stm32f405rg.html) says that full duplex I2S is supported, but the `I2S::new_full_duplex()` method isn't enabled in Embassy.  It looks like it is gated on the `spi_v4 || spi_v5` config attributes, so maybe one of these is being set incorrectly for this chip?  (Sorry, I wasn't able to figure out how these configs get set or I would have sent a PR!)

On a slightly different topic (and maybe a better issue for `embassy_stm32`), it seems like the `mck` output pin is not required for I2S rx/full_duplex operation, but it is required by the API.  Is there a reason for this?

Nybyte, 2025-09-21:
Greetings;

The problem is that the STM32F405RG supports full duplex I2S, but the Embassy configuration isn't enabling it properly. The spi_v4 and spi_v5 config attributes are set in the Embassy STM32 chip configuration files.

For the solution you need to:

You need to modify the Embassy STM32 configuration for the STM32F405. Here's what to do:

Locate the chip configuration file: In the embassy-stm32 repository, find the file for STM32F405 in embassy-stm32/src/chips/

Add the appropriate SPI version: The STM32F405 should have spi_v4 enabled. Add this to the chip's configuration:

`// In the chip's config.rs or similar file
impl embassy_stm32::peripherals::Config for Config {
    fn config() -> embassy_stm32::peripherals::Config {
        let mut config = embassy_stm32::peripherals::Config::default();
        config.enable_spi_v4 = true;  // Or set the appropriate SPI version
        config
    }
}`

Alternatively, submit a PR to fix the configuration: The proper fix would be to ensure the STM32F405 has the correct SPI version enabled in Embassy's chip definitions

Dirbaio, 2025-09-21:
@Nybyte your suggestion makes no sense. Have you used AI to generate it?

bifurcation, 2025-09-21:
@Dirbaio - Thanks for flagging that before I went off and tried to implement it :)

FWIW: I did do a little experiment with `embassy_stm32` to see whether turning on one of these features would work.  Enabling `spi_v4` resulted in failures when building `embassy_stm32` with `-F stm32f405rg`.  With `spi_v5` enabled, `embassy_stm32` builds successfully, but my calling code still can't see `I2S::new_full_duplex` for some reason.

```
diff --git a/embassy-stm32/build.rs b/embassy-stm32/build.rs
index b5f1261..169444f 100644
--- a/embassy-stm32/build.rs
+++ b/embassy-stm32/build.rs
@@ -63,6 +63,10 @@ fn main() {
                 cfgs.enable(cfg_name);
             });
         }
+
+        // XXX(RLB) Manually enable spi_v5
+        cfgs.declare("spi_v5");
+        cfgs.declare("spi_v5_i2s");
     }
 
     for &(kind, versions) in ALL_PERIPHERAL_VERSIONS.iter() {
@@ -72,6 +76,10 @@ fn main() {
                 cfgs.declare(cfg_name);
             });
         }
+
+        // XXX(RLB) Manually enable spi_v5
+        cfgs.declare("spi_v5");
+        cfgs.declare("spi_v5_i2s");
     }
 
     // ========
```

Nybyte, 2025-09-21:
@Dirbaio Sry english is not my mother languaje but I had same issue and I did what I posted there.

bifurcation, 2025-09-21:
Taking a deeper look at this repo, it seems like [this line in stm32-data-gen/src/perimap.rs](https://github.com/embassy-rs/stm32-data/blob/f7afc3878633efe70873fb3521ea1e27b3f5bc11/stm32-data-gen/src/perimap.rs#L59) might be the right place to make a change.

bifurcation, 2025-09-22:
On further research, I am seeing that the `spi_vX` tags refer to different register sets (e.g., [spi_v2_i2s](https://github.com/embassy-rs/stm32-data/blob/f7afc3878633efe70873fb3521ea1e27b3f5bc11/data/registers/spi_v2_i2s.yaml) vs [spi_v5_i2s](https://github.com/embassy-rs/stm32-data/blob/f7afc3878633efe70873fb3521ea1e27b3f5bc11/data/registers/spi_v5_i2s.yaml).  Which means that simply enabling `spi_vX` for `stm32f405rg` is not going to be the right answer.  Which means that this is an issue for `embassy_stm32` and not this repo.  

I filed https://github.com/embassy-rs/embassy/issues/4698 to track the issue there.